karawin/karadio-addons #1: Can't get communication between karadioUCSTM32 and the radio

hcremer, 2017-05-18:
Hello Karawin,

first of all many thanks for the great programming job, you are doing!

My Karadio is up and running since a few weeks. I switched it from breadboard to soldered board to get more stability. The communication between the Karadio and a serial terminal via FTDI USB Serial Device converter works fine at 115200 Baud.

When powering up the STM32 I'm getting messages on the LCD-screen and via FTDI USB Serial Device converter at 115200 Baud, I'm getting:
Setup
sys.tzo("1")
cli.info
sys.date

When pressing the remote control numbers, the number appears on the LCD.
When manipulating the rotary encoder nothing happens (VCC connected to 3.3 V and GND to GND)

When connecting:
--ESP-12F--   --stm32
  Tx                 Rx   PA10
  Rx                 Tx   PA9
  Gnd              Gnd
(Connected to different Power supplies with 5 V for stm32 and 3.3 V for ESP-12F

nothing happens. No communication between ESP-12F and STM32 (no IP no time, no reaction to rotary encoder or remote control.

What am I doing wrong? Maybe wrong board parameters while compiling the scetch?

Best regards,
hcremer

karawin, 2017-05-19:
Stop the terminal on the usb. Txfrom esp  cannot be done to the usb and  
to the STM32 at the same time. Electrical interference.

hcremer, 2017-05-19:
Thank you very much for answering so quickly!

Tried it with only TX -> RX PA10, RX -> TX PA9 and GND -> GND connected. No Terminal or anything else in between.
Not getting any messages on LCD or reactions to remote control or rotary encoder on esp-12f.

Tried to fire up first radio then stm32 and also tried to fire up stm32 and radio second, ...

No luck.

How did you upload the software to the stm32 after compiling. I'm working on linux. After flashing the firmware generic_boot20_pc13.bin, this should be possible via the usb-connector of the stm32 when putting a 1.8 kOhm resistor in parallel to R10. But I only can upload firmware and compiled program via serial (FTDI connected to PA 9 and 10).

Any ideas?

Best regards,
hcremer

karawin, 2017-05-19:
Is it for karadioUCSTM32?
Be sure to set the esp at 115200b/s
What is displayed on the lcd?

I don't know how to upload under linux. I am using the arduino ide.
Once the bootloader is flashed with the serial (Serial1), I flash the 
compiled ino with the usb interface Serial). (Why adding a resistor?)
See http://wiki.stm32duino.com/index.php?title=Uploading_a_sketch
http://www.stm32duino.com/viewtopic.php?f=32&t=413

karawin, 2017-05-19:
Check your proto with http://karadio.karawin.fr/images/KaradioUCSTM32l.jpg

hcremer, 2017-05-21:
Thank you very much for your help!
I assume the problem was loading up the scetch with the FTDI under linux. This maybe has overwritten the bootloader.
During the weekend I got access to a windows box and was able to upload the scetch via the usb-port.
Now the thing is working fine :-) and I even got a serial device under linux.
Next thing will be that I need to find or create a nice housing for my new radio ;-)
Best regards and thanks again for the great software!
hcremer

closed 2017-05-21 by hcremer