karawin/karadio-addons #2: Supply voltage usb or 5V pin

djuscha, 2017-06-30:
If 5V is fed via usb then STM32 is working normally, all commands through encoder or IR are going fast and immediately. If I feed through the pin 5V then it works slowly for a long time thinks for some reason ..
why is that? Use the same voltage supply

karawin, 2017-06-30:
May be gnd is missing and with usb it is connected by the usb itself?

djuscha, 2017-06-30:
No, Minus I have connected to GND pin

karawin, 2017-06-30:
You are right, a problem is discovered but it is a problem only with a control panel and an addon.
If you don't have a control panel, be sure to gnd the A0 input.
I will release a 1.3.3 in one hour max.
Thanks

djuscha, 2017-06-30:
ok:) Thanks  
A0 is on GND, have no buttons

closed 2017-07-31 by karawin